TheRemote/Ubuntu-Server-raspi4-unofficial #95: Real-Time Kernel

Dangield, 2019-12-16:
I tried changing a kernel to a real-time one and the raspberry won't even start. I connected it directly to the monitor and all I get is a dark screen, not even a single message. Is there something that I could do to make it work on this system? Are you by any chance going to create a release with real-time kernel?

I used the following instruction for kernel compilation and instalation:
https://lemariva.com/blog/2019/09/raspberry-pi-4b-preempt-rt-kernel-419y-performance-test

TheRemote, 2019-12-21:
This is probably a little too bleeding edge for this project.  It's still getting constant fixes as we speak here: https://github.com/raspberrypi/linux/issues/2943

closed 2019-12-21 by TheRemote

guysoft, 2019-12-30:
Hey, you can try my exported kernel package and see if that works for you:
https://github.com/guysoft/RealtimePi/releases/tag/0.4.0