TheRemote/Ubuntu-Server-raspi4-unofficial #96: HDMI blank screen after login

alezzand, 2019-12-22:
I made a successful installation (server install and full Ubuntu Desktop), and now I have an Ubuntu desktop login screen ok. But after typing my credentials and doing the login, I have a blank screen (I have an HDMI monitor attached to my RPI4 1Gb, and an official USB-C power supply). 
I tried to access with a remote desktop (VNC), but the result is a similar blank screen.
Although I don't think this is the case, I made every possible combination of config.txt changes, in the video sections, without result.
With raspi-config I increased the GPU memory, but this too didn't help.

TheRemote, 2019-12-22:
Hey alezzand,

Which desktop metapackage did you install?  Was it ubuntu-desktop-full or was it kubuntu or another variant?

I'd be curious if you download the master image with the desktop included if you are still experiencing issues as well.

Can you SSH in and grab the /var/log/syslog file?  It should have some clues as to what is happening!

alezzand, 2019-12-23:
Hi @TheRemote, thanks for our answer. 
My installed desktop was ubuntu.

Now I completely reinstalled all (always following your directions in https://jamesachambers.com/raspberry-pi-4-ubuntu-server-desktop-18-04-3-image-unofficial/). So I installed the master image with the desktop, and I'm having the same issues.
Here in this file a tail from the syslog:

[UbuntuSyslog.txt](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/files/3995157/UbuntuSyslog.txt)

Note that after some seconds from logging the raspberry becomes almost completely unresponsive (very slow) from  a remote SSH too.

TheRemote, 2019-12-28:
The key part is at the very end:

Dec 23 13:28:32 ubuntu kernel: [  484.324713] Call trace:
Dec 23 13:28:32 ubuntu kernel: [  484.324721]  __switch_to+0xf0/0x158
Dec 23 13:28:32 ubuntu kernel: [  484.324727]  __schedule+0x258/0x878
Dec 23 13:28:32 ubuntu kernel: [  484.324731]  schedule+0x3c/0xa0
Dec 23 13:28:32 ubuntu kernel: [  484.324736]  __mmc_claim_host+0xb8/0x200
Dec 23 13:28:32 ubuntu kernel: [  484.324740]  mmc_get_card+0x38/0x48
Dec 23 13:28:32 ubuntu kernel: [  484.324759]  mmc_sd_detect+0x24/0x90
Dec 23 13:28:32 ubuntu kernel: [  484.324764]  mmc_rescan+0xd0/0x370
Dec 23 13:28:32 ubuntu kernel: [  484.324770]  process_one_work+0x1c4/0x420
Dec 23 13:28:32 ubuntu kernel: [  484.324779]  worker_thread+0x50/0x408
Dec 23 13:28:32 ubuntu kernel: [  484.324786]  kthread+0x108/0x138
Dec 23 13:28:32 ubuntu kernel: [  484.324792]  ret_from_fork+0x10/0x1c

That is a mmc / SD card crash.  It's usually caused by a bad SD overclock or the SD card has gone bad.  Do you have another one to try it with?

Napoleon-BlownApart, 2020-02-14:
Have a look at this post:  https://askubuntu.com/questions/339593/missing-icons-for-headless-xfce-session-via-vnc , which solved my problem.

In my case, I was able to authenticate, but ended up with a totally blank desktop and a fat mouse cursor.

Napoleon-BlownApart, 2020-02-15:
I used `TigerVNC` in the end and now have a unity like desktop.

TheRemote, 2020-02-25:
These should be resolved in v28.  If you're still having trouble definitely open an issue!

closed 2020-02-25 by TheRemote