TheRemote/Ubuntu-Server-raspi4-unofficial #97: I have overclock but rasperi pi 4 not work

soresker, 2019-12-22:
Hi,
I have v26 then xubuntu desktop installed..everything ok..but ı had overclock in config file..
over_voltage=2
arm_freq=1750
gpu_freq=600
then rasperi pi4 not working..

Thank you very much for working rasperi pi4

TheRemote, 2019-12-22:
Hey soersker,

You should be able to fix it by plugging in your SD card to another device (Linux or Windows will both work) and navigating into /boot/config.txt and reduce the clock speeds until it can boot again.

soresker, 2019-12-22:
Hi James ,
I changed into /boot/firmware/config.txt 
Changed ;
# over_voltage=2 //disable
arm_freq=1250  //change
gpu_freq=400  //change

Everything okk Thank you feedback..

closed 2019-12-22 by soresker