TheRemote/Ubuntu-Server-raspi4-unofficial #98: The WIFI 5G seems not stable on the latest ubuntu desktop

haomingLZ, 2019-12-25:
I was using ssh to connect raspberry pi4 without a screen, it seems unstable with 5g wifi and was disconnected after a couple of minutes and can not be connected automatically again.
The problem solved when I change to connect 2.4G wifi.

3wnbr1, 2019-12-28:
Have you stumbled upon the same issue on Raspbian ? 5Ghz Wifi is known to be very picky especially at long range.

TheRemote, 2019-12-28:
It's also possible that you are in a different country than the US and need to update your regulatory country (5Ghz needs this to know the right bands to use).  

This can be changed by editing /etc/default/crda and changing US to the code for your country.

TheRemote, 2020-02-25:
If anyone is having issues still on v28 please open an issue!

closed 2020-02-25 by TheRemote

mcl21014, 2020-05-05:
I have WiFI issues as well, and noticed turning off power saving for WiFi somehow toggles on airplane mode, thus making it useless for troubleshooting, sadly. I've opened a new ticket.