TheRemote/Ubuntu-Server-raspi4-unofficial #99: (v26) Boot-up process hangs up at "Reached target Encrypted Volumes"

pseudofunizer, 2019-12-26:
After running the update and successfully booting up once, I thought everything was going smoothly, but on I rebooted again and now I am just getting stuck on "Reached target Encrypted Volumes" (I don't actually have any encrypted volumes at all, so that's an odd error.)

Been trying to look up a solution to that and apparently there's a multitude of possible explanations to the cause of that; starting from a kernel bug. 

https://www.reddit.com/r/openSUSE/comments/5y6859/boot_hangs_at_reached_target_encrypted_volumes/
https://forums.opensuse.org/showthread.php/533023-Boot-freezes-after-encrypted-volumes-target-after-zypper-up
https://askubuntu.com/questions/1116395/ubuntu-18-10-suddenly-wont-start-anymore?noredirect=1&lq=1
https://unix.stackexchange.com/questions/53820/systemd-stuck-on-mounting-an-encrypted-volume

Any ideas what happened and where I should start the troubleshooting from? 

As mentioned, I don't have any encrypted volumes nor anything attached to the USB ports apart from the keyboard and the mouse, if that helps at all. All help kindly appreciated, keep up the good dev work.

TheRemote, 2019-12-28:
Hey pseudo,

That is a strange one.  Can you recover your /var/log/syslog file by connecting your storage to a PC?

pseudofunizer, 2019-12-31:
Yes, I've got it here. Would you like the entire thing or sis there something in particular I should look for? It's ~200kb so I'd like to send it to you in private or alternatively scrub any possible "too public" identifiers out from it first.. 

There might also be a lot of useless stuff in the varlog, I think I'll just revert to the backup copy I made just before I updated to the v26, and retry running the update. I've got the varlog saved of course, so drop me a line with an email or other contact info and I'll send you the varlog. 

Thanks!

tarsij, 2020-01-04:
Hey guys,

Looks like I have a similar problem.
After updating to v26 the bootup process hangs. I get three more lines after the Reached target Local Encrypted Volumes though:
- Started Flush Journal to Persistent Storage
- Activated swap /swapfile
- Reached target Swap
My system never booted up fully on v26 either

I use the SD Card only to boot, the system is on an SSD

Have you found a solution for your issue?

Cheers

pseudofunizer, 2020-01-08:
Hi there!

Problem solved -- kind of. I reverted back to the pre-v26-build that I had, just rewrote that over the SD card and began re-tackling the update.

Before running the v26 updater, I ran all the apt updates+upgrades that were available as Ubuntu PPA and included a kernel update to 4.19.83-v8-22 , which was the version I was running before I ran the v26 updater script. Rebooted with that a couple of times, all was well.

Then, I proceeded with the actual v26 update, which again did the same thing that it did on the first boot, which was to squish the resolution to 1024x768 (am using a 1920x1200 IPS panel that has a HDMI to DVI conversion in it) - It was not possible to change the resolution to anything higher than 1024x768.

This time however, after re-inserting my pre-existing tweaks to the ```/boot/firmware/config.txt``` (only the overclocking lines), I got the rainbow screen of death.

```
[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
arm_64bit=1
over_voltage=6
arm_freq=2000
#gpu_freq=600
#core_freq=600
```

^^ Those two lines that I've hashtagged out are those that I've have had enabled with all previous builds, however in v26 they were apparently causing issues, so I disabled them for now.

What I also did is I changed the ```max_framebuffers``` value to ```1```. From what I've understood about it, it should basically limit the number of displays to 1, which is what I'm using. [ https://www.raspberrypi.org/forums/viewtopic.php?t=245789 ]

For now, all's running a-OK and the monitor resolution is also in 1920x1200.

What my overall conclusion about this would be, is that there might be something underlying with the Videocore drivers; especially when it comes to i.e. tweaking the GPU frequency.

So, as for the boot-up hang that I had with v26, I had to first run all possible apt update + apt upgrade stuff that were available, then install the v26.

@tarsij - did you make any changes to either the cmdline.txt or the config.txt under ```/boot/firmware/``` ? I'd recommend re-flashing from a backup (which I hope you have), running all possible Ubuntu PPA updates BEFORE running the updater script ( the one to that Kernel version which is closer to v26) - this might be an issue with your cmdfile.txt - double check the boot device order from there. Try getting the v26 up and running on just a MicroSD first and then implement the external stuff, such as system on an SSD. 

Since for me, during my previous v26 update attempt, the freeze-up on boot appeared out of the blue after a few reboot rounds, I will still keep an eye on this issue and consider it as "working right now, but the next boot might bring me back to square one".

pseudofunizer, 2020-01-08:
Actually @tarsij , now that I think of the oddities, or rather the boot freezing up before the desktop appears, I'm wondering if it could actually be related to the GPU freezing up - as in we were getting these hang-ups on certain parts of the boot-up process that's being displayed in textmode, while desktop might've already been loading (and hanging up). A glitch in the Videocore/GPU handling and there we go ... 

You might want to try running ```sudo raspi-config``` before your next update attempt, and choosing the "boot to CLI" (Command Line Interface) rather than desktop. From what I've read and understood, the Videocore 6 drivers still are more-or-less unstable to begin with. The problem could easily be with the GPU/its drivers-and/or-desktop handling (I'm booting straight to xubuntu). This suspicion arose from the wonkiness of available resolutions after first boot into v26 successfully and subsequent tweaking that was needed on behalf of the GPU freq, it would indicate just that.

But, then again, what do I know. :D Raspberry Pi 4B, at this stage of development, is almost like a recipe for getting all these weird quirks and whatnot, so always remember to make and keep that backup copy at hand! Murphy's Law ...

tarsij, 2020-01-10:
Hi @pseudofunizer ,

Thanks a mil for the relpy!

I haven't tried out your suggestions yet but it is highly possible that I had the same rootcause.
I haven't done an upgrade before the update to v26 and I also have my pi overclocked (gpu_freq=750).

Out of curiosity I'll try with a fresh non-overclocked v26 image to see if it's working :).

btw I was luckily able to revert to v25 (I use timeshift) so my system wasn't lost :).

tarsij, 2020-01-10:
@pseudofunizer ,

You mentioned that you have horizontal tearing when watching a video in a browser.
Don't you have the same tearing problem when you watch movies? If not may I ask you which player do you use and which output driver?

pseudofunizer, 2020-01-19:
Okay, so just today I managed to replicate this problem running the v26 with all the latest updates installed by going back to tweak the `/boot/firmware/config.txt` and adding the `gpu_freq` overclocking switch back in there. 

The next reboot then got stuck in the "Reached target Encrypted Volumes"-part again. Edited the `gpu_freq` line out of the `config.txt` and it's working again. To summarize: **The current v26 version is not very fond of meddling with the GPU frequency overclocking settings. CPU overclocking works just fine, though, all the way to 2,1GHz.**

@tarsij As a matter of fact, yes - I do have tearing issues with other video players as well. Most often the tearing happens when you're playing back video in full-screen mode; I would think that the video tearing issue has something to do on Xubuntu's / Linux's video handling, how the drivers have been coded, esp. for VideoCore VI, etc. -- if you look up "linux video tearing", you can find years worth of woes under that topic.

pseudofunizer, 2020-01-19:
Some additional info to my previous post; apparently if `rpi-update` has been used (when running a Raspbian distro), it might've disabled the GPU acceleration altogether as for now since RPi4B stores the firmware to one of the EEPROM's, unlike in i.e. Pi 3B+ did. Will likely have to probe into the firmware issue from that end, although the GPU overclocking shouldn't be as big of a deal as being able to CPU overclock.

https://www.raspberrypi.org/forums/viewtopic.php?p=1595609#p1595609

pseudofunizer, 2020-02-09:
Yippee! Good news on the video tearing & other woes ( especially @tarsij - please read! )

I had been living under the impression that my RPi 4B was up to date as for the firmware/eeprom, as I was updating my other unit with Raspbian in it, I swapped in my Raspbian build to the unit running Ubuntu Server, and ran the rpi-eeprom update in it just in case -- sure enough, it wasn't the latest version and hence was updated. 

After the EEPROM update, the video tearing issue pretty much gone altogether (at least in Xubuntu)! Overall, the performance improvements are significant in comparison to before the update. 

However, before this firmware update, I've had a persisting error message with apt upgrade considering `flash-kernel` -- perhaps this is due to something conflicting with @TheRemote 's fork and the official Ubuntu packages colliding with each other? Anybody got any clues on where to begin?

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up flash-kernel (3.98ubuntu11~18.04.1) ...
Using DTB: bcm2711-rpi-4-b.dtb
Couldn't find DTB bcm2711-rpi-4-b.dtb on the following paths: /etc/flash-kernel/dtbs /usr/lib/linux-image-4.15.0-1054-raspi2 /lib/firmware/4.15.0-1054-raspi2/device-tree/
Installing  into /boot/dtbs/4.15.0-1054-raspi2/./bcm2711-rpi-4-b.dtb
cp: cannot stat '': No such file or directory
dpkg: error processing package flash-kernel (--configure):
 installed flash-kernel package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 flash-kernel
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

@TheRemote , any idaes on that one? Otherwise everything is running smoother than any other Linux distro on the entire RPi4B, so, once again, thank you for the great work! :-)

tarsij, 2020-02-09:
Hi @pseudofunizer,

Thanks a mill for the info, I'm going to try what you said :).

Regards the flash-kernel issue it looks like the file it is looking for is in a different folder: /boot/firmware/bcm2711-rpi-4-b.dtb
What I'd try is to check if the flash-kernel paths can be configured or worst case copy the file to one of the paths it is looking into. 

+ I run in another issue after apt upgrade: the opengl driver died. when I run the glxgears -info the driver reported wasn't the V3D one (maybe it was just wrong timing for the upgrade, wil recheck later)

Regards

tarsij, 2020-02-17:
FYI. Looks like the OpenGL issue was temporary so you can discard that issue.
Related the flash-kernel I've decided to 'apt-mark hold' it. Felt safer

Regards

TheRemote, 2020-02-25:
Great stuff!  I have fixed the flash-kernel issue in v28.

closed 2020-02-25 by TheRemote