TheRemote/Ubuntu-Server-raspi4-unofficial #103: /opt/vc/ missing

Bridgee, 2020-01-09:
I'm trying to cmake a ROS package which requires certain system dependencies of Raspberry Pi, namely, mmalutil_LIBS, vcos, bcm_host, and vchostif. However, I can not find them in the current system.

Possible Reference: https://github.com/raspberrypi/userland
https://unix.stackexchange.com/questions/284469/build-linux-for-a-raspberry-pi-without-opt-vc

letmejoin, 2020-01-17:
I find that too, "3D video hardware acceleration support via vc4-fkms-v3d stack. Videos / games are giving very high and smooth FPS " doesn't need vc?
Hoping add /opt/vc/ , thanks!

TheRemote, 2020-01-20:
The full available Raspberry Pi userland and all libraries are built and installed on the image.

However, here's the rub:  The Raspberry Pi is actually not completely open source (especially the Broadcom pieces) and there are a few vc utilities that are completely closed source and not available to build.

Because of this some of these libraries are not available anywhere on 64 bit simply because Broadcom won't give them to us.  If these binaries are available somewhere that you have found for ARM64 I would *love* to be wrong.

The Raspberry Pi kernel team reaffirmed as late as September 2019 here that no 64 bit userland is planned still (scroll to the very bottom): https://github.com/raspberrypi/firmware/issues/550

This means not only can we not build some libraries your projects are depending on, but they won't even PROVIDE us with a closed source 64 bit binary and have no plans to do so.

This is obviously silly.  The CURRENT Raspberry Pi is available with 4 GB of RAM which is the maximum 32 bit is going to be able to do.  For them to say they don't even plan to give us 64 bit userland is absolutely clueless when we've already hit the limitations of 32 bit in the CURRENT Raspberry Pi.

All I can do is encourage you to leave them some feedback and say this is sorely needed for progress to continue on the Pi!

closed 2020-01-20 by TheRemote

TheRemote, 2020-01-20:
To clarify, on a 32 bit image you can just "borrow" Raspbian's pre-built kernels but since there is no 64 bit Raspbian we can't do the same for 64 bit so 64 bit versions of some of those libraries are not available.

tmastersmart, 2022-07-21:
can someone build a installer to install these on Raspbian Linux 11

TheRemote, 2022-07-21:
It took me a minute to remember what this even was but the situation has changed since I wrote this.  They actually do have 64 bit userland now.  https://launchpad.net/ubuntu/+source/raspberrypi-userland

Since it builds for 64 bit now (there's a 64 bit version of Raspberry Pi OS) this is all available now.  You can even install it on other operating systems like this: http://dev1galaxy.org/viewtopic.php?id=2967

On Raspbian I would think these would already be present but if not you should be able to follow those same instructions to install them.