TheRemote/Ubuntu-Server-raspi4-unofficial #107: Minor USB dongle instability

t206, 2020-01-19:
I'm using a wireless Logitech USB keyboard and mouse via a single dongle and noticed that one or both of the devices become non-responsive occasionally. Initially, I was able to restore functionality by rebooting the Pi 4 but later found that pulling the dongle out and reinserting it restores functionality almost all of the time. I've replaced the batteries in both devices but this hasn't improved the situation. I have not experienced random dropouts with Raspian or Ubuntu Mate, the other OS's I occasionally run on the Pi 4.

deltasierra-ufc, 2020-01-20:
I recently tried using an IOGear GUS402 4-to-2 KV switch and ran into unresponsiveness issues in v26 with MATE.  I thought it was the KV switch, but I'm using a Logitech USB unifying receiver for my Logitech Performance MX wireless mouse and now I'm wondering if that's what is causing the issue.  I'll try a wired USB mouse sometime and see if it's a mouse or KV switch problem.  From the reviews I've read, this KV switch model has worked for tons of different Linux users, although it just takes oneself having a different setup than anyone else to run into snags.
Using a USB wired keyboard, BTW, which doesn't appear to become unresponsive although hard to tell because the entire desktop interface becomes unresponsive at times.

t206, 2020-02-08:
Update: I haven't seen the USB keyboard or mouse drop off recently. This may have been fixed by a recent patch. I can't be sure.

closed 2020-02-25 by TheRemote