TheRemote/Ubuntu-Server-raspi4-unofficial #111: (V27) System is stuck in purple screen after password entry in GUI

waldekmi, 2020-02-06:
I tried Desktop version (no changes from my side), system boots to Ubuntu Desktop (gdm3). After password entry system is stuck in purple screen with mouse cursor on it. Cursor cannot be moved.
I also tried Server version, followed recommendations from "Raspberry Pi 4 Ubuntu Server.." post.
Note: I noticed multiple errors like:
“Errors we encountered while processing:
 flash-kernel
Sub-process /usr/bin/dpkg returned an errorcode (1)” 
during Updater.sh run
Installed Ubuntu-desktop manually with the same final purple screen outcome.
Only difference here is that I could move the mouse and after Ctrl+Alt_F1 system switched to the terminal and hang completely. Power cycle was required.

tgwaste, 2020-02-11:
Same issue here.  Once logged in it just takes me to the purple gradient screen.  I can move the mouse around but thats it.
[image]

TheRemote, 2020-02-25:
Do you have any other monitors or TVs you can try with?

I've heard of issues like this before with TVs that have very large resolutions.  On these TVs you sometimes need to fiddle with /boot/config.txt.  There are a bunch of items in there to test that may help get the display working.

You should also try the latest v28 version as well since it contains tons of fixes!

pranavburnwal, 2020-02-27:
V28 has the same issue.

closed 2022-05-11 by TheRemote