TheRemote/Ubuntu-Server-raspi4-unofficial #112: V26 flash-kernel update error

t206, 2020-02-08:
I've been getting this error for about a week when running apt upgrade on V26 on my RPI 4. I can't find anything addressing a workaround for this issue. I booted from my Raspian image and ran update/upgrade there but it doesn't resolve this issue in V26. Is this a known bug?

ubuntu@ubuntu:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up flash-kernel (3.98ubuntu11~18.04.1) ...
Using DTB: bcm2711-rpi-4-b.dtb
Couldn't find DTB bcm2711-rpi-4-b.dtb on the following paths: /etc/flash-kernel/dtbs /usr/lib/linux-image- /lib/firmware//device-tree/
Installing  into /boot/dtbs//./bcm2711-rpi-4-b.dtb
cp: cannot stat '': No such file or directory
dpkg: error processing package flash-kernel (--configure):
 installed flash-kernel package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 flash-kernel
E: Sub-process /usr/bin/dpkg returned an error code (1)

chrstnwhlrt, 2020-02-10:
Same here with v27. A simple `sudo apt-get upgrade` results in 

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up flash-kernel (3.98ubuntu11~18.04.1) ...
Using DTB: bcm2711-rpi-4-b.dtb
Couldn't find DTB bcm2711-rpi-4-b.dtb on the following paths: /etc/flash-kernel/dtbs /usr/lib/linux-image- /lib/firmware//device-tree/
Installing  into /boot/dtbs//./bcm2711-rpi-4-b.dtb
cp: cannot stat '': No such file or directory
dpkg: error processing package flash-kernel (--configure):
 installed flash-kernel package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 flash-kernel
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

Even stranger: I started the Update.sh on a fresh rpi4 installation of the headless version (v27) and the script "deleted itself". The error above was the same but to re-execute the script I had to download it manually. A second execution didn't delete the script.

maxmerk, 2020-02-12:
Hi! I temporarily solved this problem by copying
bcm2711-rpi-4-b.dtb

sudo cp /boot/firmware/bcm2711-rpi-4-b.dtb /etc/flash-kernel/dtbs/

t206, 2020-02-13:
Indeed it does, worked for me as well. Thanks!

christian-schlichtherle, 2020-02-16:
Same here on V27:

```
$ ssh 192.168.71.20 dpkg -l flash-kernel
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version              Architecture Description
+++-==============-====================-============-=================================================
iF  flash-kernel   3.98ubuntu11~18.04.1 arm64        utility to make certain embedded devices bootable
```

Note that the first letter should be `h` for "Hold".

You can fix this using

    apt-mark hold flash-kernel

TheRemote, 2020-02-25:
Thanks for the reports, this has been fixed in v28!

closed 2020-02-25 by TheRemote