TheRemote/Ubuntu-Server-raspi4-unofficial #115: Kernel update to 4.19.105? / Chaos+its fix after running "flash-kernel" inside Ubuntu

pseudofunizer, 2020-02-23:
I began probing for a way to update the kernel v.4.19.105 as it's now listed as a mandatory and urgent update, at least according to LWN.net's article from a few days ago. 

https://lwn.net/Articles/812876/
**"All users of the 4.19 kernel series must upgrade."**

I bit the bullet even though I had no clear idea what would happen when I did so, and nonetheless installed the `flash-kernel` apt package that I had troubles with in the past and had hence uninstalled it completely. 

I had been searching for possible solutions for the error messages it gave earlier on when it attempted to ram itself through, and the reason for that was that it wasn't able to find the bcm packages, so I copied them from another directory location to where they were needed according to instructions I found from Raspberry Foundation's forums.

Well, sure enough, beginning from the boot sequence, everything was screwed up. After nearly losing my mind, I found out that `flash-kernel` had added an outdated / unwanted line into the `config.txt` file, that's on the MicroSD jump-boot card:

`dtoverlay=vc4-kms-v3d`

_**Remove that line.** AFAIK, it's only intended to be used in Raspberry Pi 3(B+)'s, and if it's enabled in Pi4B's, it'll screw everything up, as it's related to VideoCore IV, whereas the RPi4B has VideoCore VI. Or at least in my case it did the trick and everything was back to normal with the next boot._ You learn something new every day, whew!

Now, the issue is that although everything's working again, I'm still stuck with the 4.19.97-v8-27 version of the kernel, things _should_ run on a 4.19.105 kernel now, but I'm wondering how on earth I'll get there or if it's even advisable? 

The `rpi-update` util can be installed on other OS's/distros than just Raspbian, at least according to the utility's GitHub page. (See: https://github.com/Hexxeh/rpi-update / Installing)

Any last words of advice before I start probing that realm to see what happens? :-) Thanks a lot, and I hope @TheRemote keeps up the good work. Cheers!

pseudofunizer, 2020-02-23:
Installed the ´rpi-update´as instructed on the "Installing under other OSes" section of the git page ( https://github.com/Hexxeh/rpi-update ). Ran it with no problems and everything seems to be working except that the kernel version I'm being shown is still the same-old *4.19.97-v8-27* and _not_ *4.19.105*, although the ´rpi-update´ now says that my firmware is up to date ? Which is odd. It would imply that ´rpi-update´ either has updated some kernel modules, but it's not showing on the version numbering, or it's being fooled by its own version check function.

Anyway, all help on this issue is kindly appreciated. Otherwise all's well for now. Keep on truckin'!

pseudofunizer, 2020-02-23:
OK, a small status update - and this time some more good news _(or rather, so far so good :D )_ 

After running `rpi-update` on the Ubuntu Server, I re-enabled from `/boot/firmware/config.txt` that line for the graphics overlay:

`dtoverlay=vc4-kms-v3d`

... And everything's working a-OK. I noticed that the hardware acceleration for graphics had indeed been tossed aside when that option wasn't in use, causing hardware-accelerated video (such as Youtube videos) to become an issue. 

To summarize at this point: everything seems to be working great. Video playback is smooth, no screen tearing, 3D acceleration working as well. The only thing that is still a question mark is why my kernel, according to inxi, still seems to be at 4.19.97-v8-27 ... Oh well!

TheRemote, 2020-02-25:
Hey again pseudo,

Thanks as always for the incredibly detailed and helpful information you always post!

Here is how I switch the kernel when I run the updater:

    echo "Updating kernel and modules ..."
    export KERNEL_VERSION="$(ls updates/rootfs/lib/modules)"
    sudo depmod "${KERNEL_VERSION}"

    # Create kernel and component symlinks
    sudo rm -rf /boot/initrd.img
    sudo rm -rf /boot/vmlinux
    sudo rm -rf /boot/System.map
    sudo rm -rf /boot/Module.symvers
    sudo rm -rf /boot/config
    sudo ln -s /boot/initrd.img-"${KERNEL_VERSION}" /boot/initrd.img
    sudo ln -s /boot/vmlinux-"${KERNEL_VERSION}" /boot/vmlinux
    sudo ln -s /boot/System.map-"${KERNEL_VERSION}" /boot/System.map
    sudo ln -s /boot/Module.symvers-"${KERNEL_VERSION}" /boot/Module.symvers
    sudo ln -s /boot/config-"${KERNEL_VERSION}" /boot/config

    # Create kernel header symlink
    sudo rm -rf /lib/modules/"${KERNEL_VERSION}"/build 
    sudo rm -rf /lib/modules/"${KERNEL_VERSION}"/source
    sudo ln -s /usr/src/"${KERNEL_VERSION}"/ /lib/modules/"${KERNEL_VERSION}"/build
    sudo ln -s /usr/src/"${KERNEL_VERSION}"/ /lib/modules/"${KERNEL_VERSION}"/source

    # Call update-initramfs to finish kernel setup
    sha1sum=$(sha1sum /boot/vmlinux-"${KERNEL_VERSION}")
    echo "$sha1sum  /boot/vmlinux-${KERNEL_VERSION}" | sudo tee /var/lib/initramfs-tools/"${KERNEL_VERSION}" >/dev/null;
    sudo update-initramfs -k "${KERNEL_VERSION}" -u

I think the key parts for you is to use sudo depmod "${KERNEL_VERSION}" and sudo update-initramfs -k "${KERNEL_VERSION}" -u if rpi-update has taken care of everything else!

TheRemote, 2020-02-25:
Kernel has been updated in v28, thanks!

closed 2020-02-25 by TheRemote

pseudofunizer, 2020-03-04:
Thanks so much, James! The update went without much further problems; although I did had to revert my `cmdline.txt` and `config.txt` from the backups I created just before the update, JIC *(and, as instructed)*.

*_(As a side note: I could perhaps help you in writing an automated backup / revert-to-script for the `Updater.sh` in regards of the config and the cmdline? I've noticed over time that those two files are so essential to the entire update process that it's very easy for things to go awry when they are not the way they were before the update. In fact, a majority of people's Updater-related problems can likely be traced back to changes in those two files...)_*

A curious observation: ever since expanding my system to a jump-booted SSD, I've noticed that all updates want to stick themselves into the last known partition of the `/dev/sda` drive. In this case, I've made a swap partition for testing purposes, and for some reason, a majority of these kernel-related updates (regardless of your updater script, btw) want to stick themselves into the swap partition; being `/dev/sda3` in this case. That's not at all a problem with the Updater script, but rather somewhere deeper inside Linux's update handling and so forth.

SO, an oddity - but at least for now all the sysinfo details (`inxi` and such) at least _show_ that I'm running `Kernel v. 4.19.105-v8-28`, everything seems to be in working order and so forth ... So far, so good! :)

@TheRemote If you take in any donations for the hard and unrewarding work that you do, please let me know if there's some monetary way to at least buy you a couple of virtual drinks/beer/coffees for your respectable effort in bringing at least one proper distro in working order for the RPi4B. Hats off to you, Captain, and keep up the good work. o7