TheRemote/Ubuntu-Server-raspi4-unofficial #118: 5Ghz WiFi connection mostly fails to connect and stay stable (v28)

mcl21014, 2020-05-05:
Hello,
I haven't noticed this issue on Raspbian, even after running "sudo rpi-update" and grabbing experimental firmware. I have tried forgetting the network, and I am uncertain what files or packages need to be purged and reinstalled to workaround this. 5GHz WiFi worked fine for a few weeks, now it seems to drop. The Pi 4 is in the same room a few feet away from the gateway. Everything on the machine is updated. Please leave a comment to let me know any logs you want to look at. Said WiFi network is encrypted with WPA2 Personal. I also have a 2.4GHz network I've connected to as a workaround. There are also public Xfinity hotspots in case that fails. I've read and seen this is not a new issue. I am willing to help fix said issue if needed, of course. I first thought it was my DNS servers (cloudflare) but reverting didn't fix anything. WiFi will try and try to connect, and ocassionally it will succeed, and stay stable for like 30 seconds to a minute and drop, then try to reconnect again and again and fail 3/4 of the time. I've tried to disable WiFi power saving in Settings, and via a config file, but disabling via the GUI under Settings > Power kicks in airplane mode, which doesn't help. Another bug?

t206, 2020-05-05:
I have the same issue on 5GHz. It will connect for a short time and then disconnect so I have reverted to using the 2.4 GHz network which seems quire stable.

mcl21014, 2020-05-05:
So it's not just me. I read a comment by the dev in a previous issue ticket for anyone still experiencing the issue to bring it up. Anyone else, don't hesitate to mention issues. I check my email every day for adult-like stuff (bills, post office digests, etc.) so if the dev needs anything, I'll help quickly.

deltasierra-ufc, 2020-05-05:
I’m having issues!  Glad to see it’s not just me. *Phew!*

   V/R,
Derek Swanson

splang, 2020-05-05:
> I’m having issues! Glad to see it’s not just me. *Phew!*
>
>
>
> Derek Swanson

Set wlan0 'Power Management:off' on boot - I do it via crontab @reboot.

using command :-

sudo iw wlan0 set power_save off

Try it and see.

mcl21014, 2020-05-05:
Messing with that command unfortunately did not work around the issue. Also, Settings still shows power management on for WiFi.

mcl21014, 2020-05-06:
Funny,. I just purged gnome-power-manager and the issue is gone for now. Wifi immediately reconnected to 5GHz.

StuartIanNaylor, 2020-05-06:
Anyone disabled onboard and run with a dongle for comparison?

mcl21014, 2020-05-06:
Disregard what I said. It still appears botched. I have no clue why. And why only 5GHz.

lukelu0520, 2020-07-31:
Got similar but not the same issue. Our raspberry pi 4's 5G wifi gives us about **60Mbps/35Mbps/110ms** for download / upload / latency.
However, other desktops connected to the same wifi give us about **200Mbsp/35Mbps/25ms** for download / upload / latency.

We were wondering why there are big difference on download and latency while upload is fine.

Adding an antenna could improve download and latency ? Hope to hear your expertise.

Thanks
-- Luke

TheRemote, 2022-05-11:
Make sure the country code is set or it will be on the wrong channels, that should take care of it!

closed 2022-05-11 by TheRemote