TheRemote/Ubuntu-Server-raspi4-unofficial #122: Any plan to update and add support for Ubuntu 20.04

xlazom00, 2020-05-11:
Any plan to update and add support for Ubuntu 20.04
it will be nice to have original raspberry pi kernel + with ubuntu 20.04

t206, 2020-05-11:
You may want to try the new Ubuntu 20.04 server image that became available a couple of weeks ago. Beware that it installs kernel version 5.04.x which does run on the Rapsberry Pi 4 in 64-bit mode but is very unstable. I gave it a try this weekend along with the Ubuntu Mate desktop and had troubles keeping the Pi running for more than an hour or so. It just seems to lock up, the green and red LEDs go out completely, and a power cycle is needed to get the Pi to boot up again. I also had troubles logging out - the video would disappear with just a flashing cursor in the upper left corner of the screen.

You could also try running the distribution upgrade to 19.10. After having a few problems installing some packages, I was able to successfully get all of them installed. I've been running it for two days so far, and it seems to be very stable. Cautionary note that wireguard is still not back-ported completely to 19.10 (it works OK with kernel 4.19.105 v8 but not with kernel 5.3. My understanding is that it does work with kernel 5.4 but that kernel is part of the unstable 20.04 release.

I've also had troubles going beyond kernel 4.19.105 for the 64-bit v8 version. I tried 4.19.113 and 4.19.115, both of which locked up my Pi so I backed down to 4.19.105 and am holding there. ALWAYS backup your boot volume before trying to go beyond what James supports here.

xlazom00, 2020-05-11:
@t206 I would like to try 5.4 kernel as it looks like raspberry pi is upgrading :) https://github.com/raspberrypi/firmware/commits/master

qwertyuser1983, 2020-05-16:
I use this distro because of Canonical adamant in not implementing the raspi-config, and Argon Forty hasn't provided ubuntu script, and several of my old projects depend on it. 

If you want to update to 20.04, please, please, keep the raspi-config supported. 

(´｡• ᵕ •｡`) ♡

t206, 2020-05-16:
I have already tried installing Ubuntu server 20.04 with the 5.4 kernel and found it to be unstable on the Raspi 4. However, I saved off the full install and will periodically retry it as progress is made. James’s v28 build uses the 4.19.105 kernel and is very stable after upgrading to 19.10. They are quite a few updates coming out under 19.10 so I’ll try upgrading to 20.04 in the future. I did try upgrading v28 to 20.04 at first but it failed with many package dependencies not being met. Bottom line is I think it’s a bit too early for 20.04.

TheRemote, 2020-07-17:
Hey guys,

I'm about to attempt to build the latest kernel.  I finally received the 8 GB Pi.  This should include some firmware updates.

The image has been pretty stable on the old kernel so I'm honestly nervous to even mess with it but since the new model isn't supported I'm going to give it a go.

I'll make sure to keep the raspi-config working no matter what we do!

deltasierra-ufc, 2020-07-17:
James,

   Awesome!  Thank you and g’luck!

   V/R,
Derek Swanson
IT Administrator
Ursa Farmers Cooperative
Office: 217-964-2116
Cell: 217-592-6468
Web: https://www.ursacoop.com<https://www.ursacoop.com/>

[cid:image001.png@01D65C36.28E36970]

The information contained in this e-mail is for the exclusive use of the intended recipient(s) and may be confidential, proprietary, and/or legally privileged. Inadvertent disclosure of this message does not constitute a waiver of any privilege. If you receive this message in error, please do not directly or indirectly use, print, copy, forward, or disclose any part of this message. Please delete this e-mail and kindly notify the sender as a courtesy. Thank you.

lukelu0520, 2020-07-31:
> Hey guys,
>
> I'm about to attempt to build the latest kernel. I finally received the 8 GB Pi. This should include some firmware updates.
>
> The image has been pretty stable on the old kernel so I'm [...]

Thanks for your dedicated efforts on Ubuntu-Mate on RP4 8GB @TheRemote !
We have experienced camera and slow wifi issue on current Ubuntu 18.04-Mate kernel on RP4 8GB. And recent firmware updates from raspberry pi community resolves these issue on 8GB version. 

Looking forward to your next coming update on Ubuntu-Mate 18.04 image !

-- Luke

t206, 2020-08-15:
Update on installing Ubuntu 20.04.1...

I've been running 19.10 for the past few months but it's no longer supported so I decided to try upgrading to 20.04 again.
The Pi 4 is now running 20.04.1 with the following exceptions:

1. ● multipathd.service                                                                             loaded failed failed
2. Kernel remains at 4.19.105 v8

The upgrade took 3 steps:

1. I ran the 19.10 GUI software updater. There were a few errors and one missing package which I successfully installed. After rebooting, I noticed the failing multipathd service and once the bootup process completed, I had a blank screen and no mouse or keyboard.
2. From a remote ssh shell, I ran James's Updater.sh script and it found a bunch of packages to update. After this completed, I rebooted and voila, I had a mouse, keyboard, and a full display.
3. I ran the 20.04 GUI software updater which installed more software (I don't know what). Rebooted one more time and it came right up running 20.04.1. I wasn't brave enough to try installing a 5.4 kernel.

So far so good. I expect this to be as stable as 18.04 and 19.10 since I still have the frozen kernel. I can't find any relevant info about the multipathd service. If anyone has ideas about this, please let us know.

TheRemote, 2020-12-06:
Today I released a tool to help set up USB booting for Ubuntu 20.04.  It's not the exact same as this but it is why a lot of people used it, so I wanted to leave it here.

I did a writeup for it at [https://jamesachambers.com/raspberry-pi-4-ubuntu-20-04-usb-mass-storage-boot-guide/](https://jamesachambers.com/raspberry-pi-4-ubuntu-20-04-usb-mass-storage-boot-guide/) and I have committed the file to make the Ubuntu drive boot the same way this image did and committed it to this repository.  It is BootFix.sh.

closed 2020-12-06 by TheRemote