TheRemote/Ubuntu-Server-raspi4-unofficial #127: Raspberry Pi CSI camera v2 module issue (almost gray pixels and fragile colors)

lukelu0520, 2020-07-20:
Dear Ubuntu-Server-raspi4-unofficial friends,

I setup camera v2 module up and use raspistill -o test.jpg to capture an image.
The preview looks great, however, the saved image is crappy -- contains almost gray pixels and fragile colors as shown in below link. I tried two camera v2 but got the same issues.

[captured image](https://drive.google.com/file/d/1txSMoyceXO-viV2yBRSfbVYTr7SQHw0a/view?usp=sharing)
[camera v2 module details](https://www.amazon.com/gp/product/B07W6THFPH/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)

Was wondering if anyone got the same issue with me. Hope to hear your expertise.

lukelu0520, 2020-07-25:
This issue is recently fixed by the latest Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux

https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=280664&p=1702011#p1702011

Will try if it works on Ubuntu 18.04.

chaitunya, 2020-08-03:
I have the same issue. As a workaround, I'm using the picamera library to take pictures:

```python
from picamera import PiCamera
cam = PiCamera()
cam.capture("foo.jpg")
```

Also, when you use the bcm2835-v4l2 module, you can use the camera act as a webcam, so you can use it in webcam apps like [Cheese](https://wiki.gnome.org/Apps/Cheese). To activate the bcm2835-v4l2 module, run this command:

```bash
sudo modprobe bcm2835-v4l2
```

closed 2020-12-06 by TheRemote