TheRemote/Ubuntu-Server-raspi4-unofficial #128: Please continue this work on 20.04

Askmewho, 2020-09-05:
Title.. we need this. Desktopify was a pain in the ass for me to setup.

flipkickmedia, 2020-09-06:
Seconded...I'm trying to get ubuntu 20 to work with a diskless boot and its proving tricky.

TheRemote, 2020-12-06:
Hey guys,

I didn't do a full blown image this time but I did make a tool (BootFix.sh in this repository) for Ubuntu 20.04 / 20.10.

Directions on how to use it are at https://jamesachambers.com/raspberry-pi-4-ubuntu-20-04-usb-mass-storage-boot-guide/ and there's some basic instructions in BootFix.sh, but it's pretty dang easy and works really well!

https://jamesachambers.com/raspberry-pi-4-ubuntu-20-04-usb-mass-storage-boot-guide/

k3nden, 2020-12-28:
James, 
I wanted to report that I was unable to use bootfix on Ubuntu server 20.04 (64 bit version).   Both server and desktop versions of 20.10 works great. Thanks

flipkickmedia, 2020-12-28:
It has been a while but I did a guide:

https://askubuntu.com/questions/1273084/raspberry-pi-diskless-cluster-using-ubuntu-20-04-1

I got fuse fs working and am using this on a daily basis now.

closed 2022-05-11 by TheRemote